kiu/lofence #1: How to re-program RN2483 with your board layout?

kevinkngo, 2019-11-27:
Hi kiu,

I really appreciate you sharing this project. This is my first IoT project where I've started to enter the world of AVRs and PCBs (which is very exciting!).

I have actually gotten your PCB design fabricated, soldered on the components you listed in your BoM and loaded the firmware you provided onto the MCU... And it works great.
I am currently trying to re-program the RN2483 soldered onto my PCB using the LoRaDevUtility ([Microchip documentation](http://ww1.microchip.com/downloads/en/devicedoc/40001847a.pdf)) and the RN UART pins you have provided. 

However, I believe that due to the way you have set out your board, it is very difficult to do this. I initially could see the RN2483 in the LoRaDevUtility (see screenshot 1), but had to continuously hold down SW1 to keep the UART connection "alive".
I then tried to update the firmware on the RN2483 but I accidentaly "bricked" the chip currently on my board because the UART connection disconnected while the firmware was updating. Now whenever I try to upload firmware onto the RN2483 now, it keeps disconnecting and does not allow me to upload any firmware (see screenshot 2). 

I am more than happy to grab another RN2483 and try again, but I'm hoping that you could provide some advice before I continue.

Do you have any recommendations about how I can re-program the RN2483 using your board layout? Have you tried uploading different firmware onto the RN2483 yourself? 
Since you manufactured so many boards, I am curious whether if you've thought about how you would update the firmware on the RN2483 on these boards (without taking it off?).

Thanks,
Kevin

[image]
[image]

kiu, 2019-11-27:
Hi Kevin,

I have to admit, I didn't think much about how to upgrade the firmware of the RN module. Or at least not in an user friendly way.
I would probably flash the AVR with a new config disabling the RX/TX pins, so they won't interfere with the RN module. Then upload the the RN firmware as you did.

If the RN module leaves bootloader mode, you could try to send the "eraseFW" command.

Hit me up on https://www.thethingsnetwork.org/community/toowoomba/post/join-us-on-slack (user: kiu).

kiu, 2019-11-29:
For the record.

Flashing a NOP version of lofence with RX/TX pins disabled (Atmel Start Config) and issuing a single RESET does the trick.

`int main(void) {
	atmel_start_init();
	_delay_ms(100);
	RN_RESET_set_level(false);
	_delay_ms(100);
	RN_RESET_set_level(true);
	_delay_ms(100);
	while (1) {
	}
}
`

closed 2019-11-29 by kiu

jpmeijers, 2021-11-03:
I was struggling a little to do the firmware upgrade. Mostly because I am not familiar with Atmel Start. So let me document what I did in case someone else stumbles upon this thread.

1. Comment out the content of the `int main(void)` function and replace it with the contents of the main function as posted above.
2. Open src/driver_init.c and comment out the lines `USART_0_initialization();` and `USART_1_initialization();`.
3. Upload to LoFence board.

Because I could not find the Microchip firmware upgrade tool from an official source, and the one I did find didn't want to work, I looked for a different way to flash the firmware. I ended up using https://github.com/SodaqMoja/RN2483FirmwareUpdater

Luckily I had a Sodaq Explorer at hand to do the update, but I had to change the firmware to program the external RN2483 and not the onboard one on the Explorer. In RN2483FirmwareUpdater.ino on line 17 I replaced `Serial2` with `Serial`. `Serial` is the UART on pin 0 and pin 1. 

Connection between the Explorer and LoFence is as follows:

| Explorer | LoFence J4 |
| -- | -- |
| 3.3V | pin 1 (3.3v) |
| D1 (TX) | pin 2 (RX) |
| D0 (RX) | pin 3 (TX) |
| GND | pin 4 (GND) |