kiu/lofence #2: missing symbols

toozie21, 2020-07-23:
Nice design!

I am new to KiCad and when I opened your project there were missing symbols for the schematic (like for the RN2483).  Are those symbols something that you drew by hand? I haven't looked at the PCB yet, but I will let you know if I see similar issues there.

kiu, 2020-07-23:
Uh, thanks for pointing that out. I accidentally gitignored the cache.lib file.
Please update and check again.

toozie21, 2020-07-23:
> Uh, thanks for pointing that out. I accidentally gitignored the cache.lib file.
> Please update and check again.

Thank you, that did solve that issue.  Though it seemed to cause a different one after I updated...
[image]

kiu, 2020-07-23:
Odd. I did a clean install of kicad on a machine where I never used kicad on before. Did a git clone and get no errors. I do get a wizard though, but that works fine.

[image]

toozie21, 2020-07-23:
Weird. What version of KiCad?

kiu, 2020-07-24:
I installed latest stable 64bit Windows Version, that is 5.1.6
https://kicad-pcb.org/download/windows/

toozie21, 2020-07-24:
> I installed latest stable 64bit Windows Version, that is 5.1.6
> https://kicad-pcb.org/download/windows/

I was using Ubuntu and it was one minor version behind.  I installed it on my Windows laptop to make it easier and it seemed to load up fine, so I think you solved the problem, thanks!!

closed 2020-07-24 by toozie21