kiu/lofence #3: Question on battery

toozie21, 2020-07-27:
Not an issue, just a question.

In the BOM you say this for the battery
`18650 Battery 3350 mAh (Not recommended!)`
But I don't see any explanation anywhere for why you don't recommend it. Did you have issues? TIA

Alex9779, 2020-08-20:
Yeah I am interested in an answer on this too...

kiu, 2020-08-20:
If I remember correctly, the battery doesn't have a security circuit. Therefore I wouldn't want to recommend it. Otherwise nothing wrong with it. Works fine for me.

toozie21, 2020-08-20:
> If I remember correctly, the battery doesn't have a security circuit. Therefore I wouldn't want to recommend it. Otherwise nothing wrong with it. Works fine for me.

That is correct. I thought that if you take an additional step, it would need the protection and probably charging circuitry. I think it is perfect as is though.

I just wanted to make sure that you didn't have one blow up on you, and we're suggesting a different brand 😁. 
 Thanks again for the awesome design.

closed 2020-08-20 by kiu