kiu/lofence #4: Question on voltage divider errata

Alex9779, 2020-08-20:
Just another question on the errate you did on the voltage divider.
Could you just explain what you meant with that?
I am thinking of rebuilding your design but also I am puzzled what the difference is as I am not really sure what you mean with "Fence+", "Fence-" and "12kV".
My fences have just one pole, when I measure them using a normal tester the tester is connected to the ground by a nail and one end goes to the fence. There is no + and - but just a +, - is ground...
What am I missing?
Thanks for your efforts in advance...

kiu, 2020-08-20:
The left side is REV B and right side is the potential improvement. The hot wire of your fence is connected at the top, bottom has the earth connection (rod/spike in the ground)

I am wondering in the errata, if its clever to switch to the design on the right, where the pcb ground and earth of the fence are decoupled by 10M.

Does that help?

[image]

Alex9779, 2020-08-21:
Yeah cool thanks, I think I will calculate that for me to see the numbers too but the changed labels helped a lot...

closed 2020-08-21 by Alex9779

frederickjh, 2023-04-10:
> Yeah cool thanks, I think I will calculate that for me to see the numbers too but the changed labels helped a lot...

@Alex9779 Thanks for posting this issue. I was wondering the same thing. Thanks to @kiu for the diagram explaining it.

Did either of you ever build and test this improved voltage divider? 

Just wondering and trying to not reinvent the wheel. 

I see one commercial product (Fence Hawk) on the market that builds the voltage divider in a piece of PVC pipe with PVC caps on the ends and stainless steel bolts in the caps to connect to the fence and ground. They then run a low voltage wire run to the detector circuit. This seems like an interesting design.

Alex9779, 2023-04-10:
I built them according to the errata and they have been working fine since then, that is the changed voltage divider and also removing the obsolete LDO circuit.