kiu/lofence #7: Issues regarding the RN2483 LoRa module and LoRa logic

Alex9779, 2020-09-17:
Taking the next step to get the projects to use I had to set up a LoRa gateway (did this already earlier and played with some other modules), learned about TTN and ChirpStack.
And I stumbled over a few things I would like to mention, no idea if you are aware of them or not.

The overall usage of LoRa does not seem to fit any guidelines, at first you use ABP for joing, that's ok but not recommended so I changed the firmware to use OTAA which then has some other things to follow like saving the join information accross power cycles. I am still working on that, I think of totally rewriting the firmware, if ready I will put it up in my fork.

Another thing is that the module uses not the EU868 specs if you do a "sys factoryRESET", you have to do a "mac reset 868" also to load the EU868 channel defaults deactivating the use of channel 3-15 (actually 7, because 8-15 are disabled by default). This has no effect if using TTN because a TTN gateway listens on all 8 channels but recommended is to use only the first 3 (0-2). I had that problem when using a gateway created with ChirpStack which just listens on channel 0-2 by default.

Last the data rate. It is left at default DR0 which is SF12BW125 it seems with no ADR while ADR is mandatory for SF11 and SF12 and based on https://avbentem.github.io/airtime-calculator/ttn/eu868/8 at DR0 only 20 messages per day are allowed and accepted by TTN but the firmware does a measurement once every hour. So if your devices use that firmware in public I would at least recommend to change it to use DR2.

Just a side note: thanks for the great project, I don't think I would have been able to get a working PCB without many prototypes and in a short time as I have...

kiu, 2020-09-17:
Thanks for sharing your insights!

> The overall usage of LoRa does not seem to fit any guidelines, at first you use ABP for joing, that's ok but not recommended so I changed the firmware to use OTAA which then has some other things to follow like saving the join information accross power cycles.

As the area I am trying to cover has really low coverage, I chose to use ABP to increase the chance of packets reaching the gateway. Didn't want to rely on additional OTAA traffic. But you are right, it is recommended and way more convenient.

> I am still working on that, I think of totally rewriting the firmware, if ready I will put it up in my fork.

Sweet.

> Another thing is that the module uses not the EU868 specs if you do a "sys factoryRESET", you have to do a "mac reset 868" also to load the EU868 channel defaults deactivating the use of channel 3-15 (actually 7, because 8-15 are disabled by default).

Hmmm, odd. I checked the Command Reference and everything points me towards that the 868 parameters are the defaults of the module. It explicitly says "The default settings consider only the three default channels (0-2), and their default duty cycle is 0.33%.", but I never verified it. Do you have more information on this?

> Last the data rate. It is left at default DR0 which is SF12BW125 it seems with no ADR while ADR is mandatory for SF11 and SF12

True, I didn't know that. https://github.com/kiu/lofence/commit/61f52703dbc37d511e447d4d42a77dffc8d7a1df

> and based on https://avbentem.github.io/airtime-calculator/ttn/eu868/8 at DR0 only 20 messages per day are allowed and accepted by TTN but the firmware does a measurement once every hour. So if your devices use that firmware in public I would at least recommend to change it to use DR2.

Also true, changed it as well.

Alex9779, 2020-09-17:
Yeah well, the module surprised me also.
The overall appearence made me also think this is plug and play but there are some drawbacks.
Trying to get OTAA working the way its meant to be, that is just one join when the device is provisioned and hopefully never again, which is considered best practice, lead into other issues.
I found an Arduino lib for the module, but this is not doing this also very well, I think of taking that and rewriting and using it for this project, most of the code should be copy paste but class init and comm must be changed for the Atmel.

Alex9779, 2020-09-17:
This is the lib I want to adapt: https://github.com/jpmeijers/RN2483-Arduino-Library

Alex9779, 2020-09-17:
BTW I checked all I wrote by directly commanding the module with a terminal app running firmware 1.0.5, that version you get if you order using the number in your BOM, no idea if the other available version does the same, reflashing seems not so easy, I found a script on Github too but I don't want to try just to prove something...

Alex9779, 2020-10-08:
Well it seems I didn't have the time (yet) to do what I wanted, I just modified the firmware to use OTA and reorganized a little.
I wrote a simple python script for initializing a fresh RN2483 module for OTA.
My 6 devices are up and running, 2 in the wild, 4 on my desk waiting for cables to connect them to the fence.
Monitoring works just fine.
Unfortunately I modified the PCBs with a KiCAD nightly build and I am not sure if I should publish that yet. I will when v6 is released and then also along with my changed firmware, so maybe January...

closed 2020-10-08 by Alex9779