kiu/lofence #14: Impulse measurement

Inrebusillis, 2023-09-19:
I have built the divider and the amp circuit and send the signal to a Dragino LSN50 V2 and V3 to make easy the LoRa part for me as I have this module from othes project,
everything working excellent BUT:
- the fence ( big bully up to 12kV is powered by voltage impulse ( about 30 ms)  every 0.8 seconds so Is not constantly all the time powered ( my unsterstand) 
- the LNS50 wake up take a measurement from the circuit and go sleep for the next 20 minutes

obviusly the probability that the impulse is done just when the module is awake is really low.

How can measure the voltage and "hold the value" till the sending module is awake again?
Or I missing something ?
thanks

jpmeijers, 2023-09-19:
Your understanding is correct.

Each time the Atmega 328p wakes up (every hour), the firmware samples the ADC for 15 seconds, at a rate of 250kHz, storing the max and min values it encounters. It will therefore measure more than 15 pulses in a row, storing the maximum sample, of the maximum pulse, and report that as the voltage.

You can have a look at how the ADC is set up here:
https://github.com/kiu/lofence/blob/9a1d09bd57b84be4a4ec193df7fd8c69c1f981a8/rev_b/firmware/LoFence/main.c#L52

Inrebusillis, 2023-09-19:
so to built it I have to use the full project.
let me ask you: I have a friends/farmer that need to 7 of these,
I am already running a TTI network.
Can I use your project? which conditions?
I would like to have a short chat with you, could you send me your tel via PN?
Thanks

Alex9779, 2023-09-19:
I tried the same about a year ago, I developed several iterations of a "sensor module" using just the divider and the opamp, one verison also used a smaller AVR MCU but I wasn't able to work reliable.
One problem is the spefic characteristics of the ADC of the ATmega 328PB of the original design.
The other is the firmware of the LSN50 which is not capapable of "waiting" a defined time frame for a function running in the background to get the max value of several impulses.
I tried to program a new firmware using the ADC of the LSN50 itself but didn't succeed.

ATM I am finalizing a V2 version with a smaller case, a different LoRa module and two 18650 or 18500 batteries, that is not fixed yet...

Alex9779, 2023-09-19:
This is how one version looked like:
[image]

And this how my current design looks like:
[image]

Alex9779, 2023-09-19:
One thing to consider is that the circuit with the OpAmp to "store" the peak and let the LSN ADC measure that is that it also uses current and from what I calculated the LSN battery would last about 2 years when measuring every 5 minutes like I do now.
And that battery is not chargeable.
This was something I wanted to accept and which also would lead to some money each year or two to keep it running...
In the end I chose to redesign the original device to have more lifetime, not to deep discharge the batteries when they run low and I dont change then right away and a different case which can be opened without dismounting it every time...

Alex9779, 2023-09-19:
@Inrebusillis Tried to send you a mail on your homepage, no idea if that worked, contact me by mail if you want to talk, think you are in  Germany too...

Inrebusillis, 2023-09-19:
thx, 
sent you a mail

Inrebusillis, 2023-09-25:
thanks,[jpmeijers](https://github.com/jpmeijers) you nailed it.
Each time the Atmega 328p wakes up (every hour), the firmware samples the ADC for 15 seconds, at a rate of 250kHz, storing the max and min values it encounters. It will therefore measure more than 15 pulses in a row, storing the maximum sample, of the maximum pulse, and report that as the voltage.

closed 2023-09-25 by Inrebusillis