kiu/lofence #15: DualOp Amp Use

ADebla, 2023-09-25:
Hello everyone,

I don't really think we could qualify this post as an issue but I need some more understanding about the hardware.

I have tried to simulate the circuit with the dual OP Amp as in the schematic.

I am actually wondering why it is the use of these twos ? I mean, as long the voltage divider gives the peak of the pulser trough the ADC (ofc, we need to hold the peak value to sample it). 

So could maybe someone explain me a bit the use of the dual op amp please ? Because I have the impression that only the voltage divider with a cirucit that hold the value feeding the adc with his negative potential to ground will be enough.

Please do not hesitate to correct me if I am misunderstanding the concept.

Kind regards,
Antoine

jpmeijers, 2023-09-26:
"Ground" in this circuit is just some arbitrary reference that remains relatively stable over the time it takes for a single pulse to pass. Because the sensor is physically connected to the fence wires, this ground value is roughly the same as the fence idle voltage, which may or may not be the same as the actual earth voltage - due to static buildup this could be different.

If you would look at the fence pulse on an oscilloscope, you will see the pulse do not only go high above the idle voltage, but also low below the idle voltage. When I compared the readings of this sensor with another fence tester, it looks like other testers report the voltage difference between the pulse peak and trough. So in other words Vmax-Vmin.

This tester measures both the positive peak voltage as well as the negative minimum voltage. This negative voltage measurement requires the second inverting opamp cirtcuit, while the maximum voltage is measured using the normal non-inverting opamp. That's why we need both opamps.

I hope this makes sense, and I hope I did not make any mistake in my description.

ADebla, 2023-09-28:
Thank a lot for the explanation ! Indeed, it does make sense now, I did work with some fence energizer but I have to say that I did not look at the negative part after the pulse.

In addition to your explanations, I understand now with the simulation does not work. In fact, I have simulated not with a pulse but with a 3.3V DC continous voltage which refer to the 12kV maximum before the voltage divider.

As it is continous, it can output and so simulate the positive side but not the negative one. I will try to simulate again and give some results here.

As a last question and for the needs of the simulation, do you have any example of the idle voltage of the fence you worked with and where did the negative part of the pulse have been gone so far ?

Kind regards,
Antoine

ADebla, 2023-09-28:
It is indeed really relevant, thanks a lot for the help and the informations. 

I think we can close the "issue" !

closed 2023-09-28 by ADebla