kiu/lofence #16: New resistor configuration in errata.txt

djorr5, 2023-10-02:
Hi, I saw this had activity starting up not that long ago and thought I might join the party ;).

In the errata.txt. OP makes mention of changing the resistor configuration from:
 [Fence+]--[10M]--[10M]--[1M]--[Measure]--[5k8]--[Fence-/GND]

to:
 [+12kV]--[10M]--[1M]--[Measure]--[5k8]--[GND]--[10M]--[Fence-]

Assuming so that the Fence- is possible earth (as in poke in an earth spike) and GND is ground referance on the PCB. However when I put the updated resistor values into falstad.com and with a 12kV input I was reading 6.28V at the opamp input. Isn't this too high for the opamp? 

Or am I completely missing something?

Sorry for making it an issue but I saw some other do it and thought this would be the best way. 

Great project by the way!!

jpmeijers, 2023-10-02:
Did you remember to move the curcuit's GND connection to between the 5k8 and 10M resistor?

"GND" is just some arbitrary chosen reference. The errata changes this reference to be isolated from the fence's negative line (for two line systems). This is good, because both the "positive" and "negative" lines will shock you if you touch them, and connecting one of these lines to GND will put the circuit and antenna at that same level - ie. the antenna will shock you just like the fence.

Maybe share a screenshot of your two simulations so we can see if there isn't a mistake in there. Circuit theory says the two voltage divider circuits should result in the same voltage between GND and Measure.

djorr5, 2023-10-03:
I did move the GND connection. But what I may not have done correctly is earthed the return wire of the fence. Here (australia) majority of fences have an earth spike and run just the one hot wire. So I was just wondering if that would affect the new layout of your voltage divider. Please see screenshot of a simulation I ran. Might not have it configured correctly, but I simulated the max voltage as a DC to see the highest level. It is set at 12kV. 

[image]

kiu, 2023-10-03:
[image]

djorr5, 2023-10-03:
Ah ok. Understood.

djorr5, 2023-10-03:
Ah ok. Understood.

closed 2023-10-03 by djorr5