kiu/lofence #17: Resistors of voltage divider for battery measurement

Alex9779, 2023-10-13:
I am wondering why those resistors have the value 232R.
Does it really matter or does this have a special reason?
Or can I take also 220R or 270R?

kiu, 2023-10-13:
As long as both have the same value its fine. The lower the value the more power is wasted. Don't know why I chose those in the past and not a few kilos.

Alex9779, 2023-10-13:
Ya ok so for the time the battery is measured current flows and is basically wasted... As far as I can see about 9mA for 500ms.
So 100k like the others used on the opamp circuit would also be fine? Should then be about 21uA.

Alex9779, 2023-10-14:
Just tested with my V2 dev devices, works like a charm, measurement are equal. so I am just using 100k now...

closed 2023-10-14 by Alex9779

jpmeijers, 2023-10-16:
It's a trade off. 

Higher value resistors mean lower power draw, but they are more susceptible to noise, causing a fluctuating or inaccurate measurement. Also with higher value resistors one need to wait a short while for the internal ADC capacitors to charge up before doing the measurement.

With lower value voltage divider resistors, the ADC cap charges up quicker, and it's less susceptible to noise, but the power draw is higher.

What one would normally do is to connect the Bat-/GND side of the divider to a low side switch (PMOS), or just to a GPIO pin. In idle mode switch the MOSFET off, pull the GPIO high, or make the GPIO high impedance. That way there is no current flowing when you are not making a measurement. Only when a measurement is done, switch the voltage divider on, wait a millisecond or two, then do the measurement, before switching it off again. In this case one can use smaller resistors to get a more accurate reading, while not wasting power in idle mode. The trade off is a higher component count (if you use a MOSFET), or a extra GPIO used.

Some info here: https://electronics.stackexchange.com/questions/546121/placement-of-mosfets-in-switchable-voltage-divider-for-voltage-measurements

Alex9779, 2023-10-16:
Thanks for further explaining, that is how it is done on the LoFence, BAT GND is on an GPIO and switched on when measuring the battery power.
I switched to 100k on my current V2 prototypes and two are in the wild now and performing pretty well, so I think I'll stick to 100k.
The measurement time is 500ms, I did not change that from the original firmware because it seemed appropriate.
Thinking about it a bit further you mentioned charging time of the ADC caps, but the imeplemtation uses ADC in free running mode and reading the value when the ADC triggers and as far as I understood the datasheet that happens when a measurement is complete, so does this really matter? If the caps charge slower does and individual measurement of the ADC not just take a little longer and using hight resistors results in fewer triggers or the ISR method during the measurement period?
But with noise you are right but it seems to have very little influence in this case...

Alex9779, 2023-10-16:
BTW the main reason I asked was because 232R is a strange value and has a bad availability at the assembler and also costs a lot more...